Bioreactor/Bioreactor_v4 #47: Noise level reduction of the stepper

lpatiny, 2017-01-09:
Seems a stepper driver could be useful to reduce the noise level.

Based on : http://ebldc.com/?p=187 this could be achieved by increasing the Switching Frequency to something between 30 to 50 khz and using the following driver :

http://www.ti.com/lit/ds/symlink/drv8811.pdf

qcabrol, 2017-01-16:
http://r6500.blogspot.ch/2014/12/fast-pwm-on-arduino-leonardo.html

Fast PWM : single slope on timer 1 can go as high as 31kHz with a 8MHz crystal MCU

lpatiny, 2017-01-16:
Control of the current seems that it can also reduce the noise and the heat of the stepper

lpatiny, 2017-01-21:
D9 seems the perfect candidate for the driver. See also https://github.com/Bioreactor/Bioreactor_v4/issues/59

qcabrol, 2017-04-12:
Stepper driver is tester and functional. Proper stepper thread to be implemented

tseesen, 2017-08-03:
Proper stepper thread has been implemented. Stepper is now functionnal on las version

closed 2017-08-03 by tseesen