Bioreactor/Bioreactor_v4 #72: Bug in PID part ?

lpatiny, 2017-01-28:
[image]

Seems strange to me that the 2 resistors are in serial no ?

qcabrol, 2017-01-30:
Problem due to HEXFET voltage polarization >4V, use a suitable FET like IPB081N06L3GATMA1
http://www.mouser.fr/ProductDetail/Infineon-Technologies/IPB081N06L3GATMA1/?qs=sGAEpiMZZMshyDBzk1%2fWi4lI00SCTc04lxzIWulRTHY=
Or even better using a suitable IGBT (more expensive though)

lpatiny, 2017-02-04:
We still have 2 problems in the current version.
1. we should connect both resistance (the one to the gate and the one to the led) directly to the microcontroller. Currently with 1kohms resistance we loose 0.8 volts ! The best is to directly connect both resistor to D6.
2. With IPB081 we are still outside specification or at least really at the edge.
[image]
Removing the resistance (disabling the control LED) helps but the IPB081 still reached 160°C when I push a little bit the heating plate. I guess we could survive with it for now but there should be a better solution.

commit 5b89b8c4 (2017-02-05): correted PID diode, 4.5 finished, closes #72, ref #47
closed 2017-02-05 by qcabrol