Bioreactor/Bioreactor_v4 #73: Shopping Cart

qcabrol, 2017-02-05:
PowerFET operating below 3V threshold with low RDS (4mOhms):
http://www.mouser.ch/ProductDetail/NXP-Semiconductors/BUK965R8-100E118/?qs=sGAEpiMZZMshyDBzk1%2fWi735hrgcpETc3k%2fKUwf3lhI%3d

qcabrol, 2017-02-15:
High Side power switch circuit with enhanced N-MOS 2n7002 (SOT23) to the ground followed by a MMBT4403 PNP triggering a 5.6V zener (SOD80) line

Zener Diode Choice
http://www.mouser.ch/ProductDetail/Vishay-Semiconductors/TZM5232B-GS08/?qs=sGAEpiMZZMtQ8nqTKtFS%2fC7KvLOwmmQC8eAIP5AcSGU%3d

2n7002 reference
http://www.mouser.ch/ProductDetail/ON-Semiconductor/2N7002LT1G/?qs=sGAEpiMZZMtqO%252bWUGLBzeAme2MkyyOCy

lpatiny, 2017-03-06:
Ok I bought the IRL2505

closed 2017-03-06 by lpatiny