Bioreactor/Bioreactor_v4 #77: Normal heating mode & flow process

apatru, 2017-02-20:
Bioreactor process: 

1) Normal heating: Stepper ON, Pump_in OFF, Pump_out OFF, Heating ON -> target temperature -> sedimentation
2) Sedimentation: Stepper OFF, Pump_in OFF, Pump_out OFF, Heating ON -> target sedimentation time -> empty
3) Emptying: Stepper OFF, Pump_in OFF, Pump_out ON, Heating ON -> target min weight -> fill
4) Filling: Stepper ON, Pump_in ON, Pump_out OFF, Heating ON -> target max weight -> normal heating

lpatiny, 2017-02-21:
And what is currently wrong ? It seems this is the way it is programmed.

apatru, 2017-02-21:
Seems like in the normal heating procedure, it switches to sedimentation before reaching target temperature. For the rest, it is right.

lpatiny, 2017-02-22:
I don't understand exactly what you need.  I just removed the heating during sedimentation and pump out I don't know if it is ok. The principle now is that each process manage itself so what I did is to turn off heating in the PID process is one of those 2 processes is active.

https://github.com/Bioreactor/Bioreactor_v4/commit/0b01dd2ba1cad69e2b19052c3d5d56dce4257434

lpatiny, 2017-02-22:
I added a faq about the 'state of the bioreactor' http://bioreactor.github.io/Bioreactor_v4/using/faq.html

The Normal mode is determined by the variable `PARAM_FILLED_TIME` and is not related to the temperature reached in the current version.