Bioreactor/Bioreactor_v4 #79: Weight measurement

apatru, 2017-02-20:
Weight measurement is jumping when stepper is ON. Shielding loading cells cables could improve that

lpatiny, 2017-02-22:
Do you have numbers with and without shielding ? You can copy / paste a `wt` result

apatru, 2017-02-22:
With stepper ON:
Current weight
-7491 = 753g
-7586 = 787g
-7600 = 792g
-7545 = 772g
-7521 = 764g
-7461 = 743g
-7461 = 743g
-7457 = 741g
-7453 = 740g
-7530 = 767g

With stepper OFF: 
Current weight
-7536 = 769g
-7538 = 770g
-7533 = 768g
-7539 = 770g
-7533 = 768g
-7539 = 770g
-7536 = 769g
-7526 = 766g
-7537 = 770g
-7530 = 767g

apatru, 2017-02-22:
no shielding has been mounted yet

lpatiny, 2017-04-04:
We circumvented the problem by checking the more stable values out of 5

closed 2017-04-04 by lpatiny