Bioreactor/Bioreactor_v4 #97: Missing Files from Solidworks CAD files

iancollmceachern, 2018-06-25:
Hello, When I clone and open the Solidworks CAD assembly I am missing many part files including:  Peristaltic Pump, Stepper 33, A Magner support v2b, A magnet support v2t, recipient, case top cover, weight sensor, a screen button.  Is there an up to date CAD export from Soldworks with all the parts?  Thank you!!

lpatiny, 2018-06-27:
Unfortunately we don't have those solidworks files. We tried to use it but is not really compatible with opensource projects so that all the final designs of the case are done in freecad.
The freecad files are fully parametric and is relatively easy to adapt to design changes.

closed 2018-06-27 by lpatiny

iancollmceachern, 2018-06-27:
Wonderful, thank you for the reply!

Where can I find the freecad files?

Thank you!

ian

lpatiny, 2018-06-28:
Everything should be in:

https://github.com/Bioreactor/Bioreactor_v4/tree/master/CAD/v4.3